STMicroelectronics/STMems_Standard_C_drivers #4: Plans to add driver for LSM9DS1

silvergasp, 2018-08-16:
Hi, 
I am interested in using the LSM9DS1 as a component in a multi-sensor IMU run on an STM32. Are there any plans to add a C driver for this component, or would you be willing to take a pull request if I were to write one myself?
Thanks

silvergasp, 2018-08-21:
I have done a bit of a look through the LinuxIIO drivers for this chip, and it seems that the driver could be created as a combination of the lsm6dsm_STdC and the lis3mdl_STdC drivers. Can you confirm if I am on the right track?

albezanc, 2018-08-22:
Hi, standard C driver for lsm9ds1 will be release soon. 
Thanks for your request.

silvergasp, 2018-08-27:
Brilliant, I look forward to hearing more

Are you able to give an ETA?

Looking at options that fit into a project timeline?

albezanc, 2018-08-31:
planned in a couple of months

silvergasp, 2018-09-04:
Brilliant, thankyou

closed 2018-09-05 by albezanc