STMicroelectronics/STMems_Standard_C_drivers #17: lis2dh12 function for calculating mG wrong implementation

IzidorM, 2018-11-23:
Function 
float lis2dh12_from_fs2_hr_to_mg(int16_t lsb)
{
  return ( (float)lsb / 16.0f ) * 1.0f;
}
is not implented right. According to datasheet page 16 where modes are described, the conversation for high res fs2 mode is 1 mg/digit, so the function doesnt have to do anything at all :) Check the other conversation functions as well, because I thing they are foo to :)

https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/4cef7a6e3ed361c48ba26772bb8f993e583037e4/lis2dh12_STdC/driver/lis2dh12_reg.c#L104

albezanc, 2018-11-23:
Hi IzidorM,

if you refer to to datasheet Rev 6, page 16, Table 10: the resolution is @ 12 bit while the driver function 
`lis2dh12_acceleration_raw_get` has a 16 bit data output.

Thanks for your clarification

closed 2018-11-23 by albezanc

IzidorM, 2018-11-23:
Ok. You are right. shift for 4 is to allign the data. My bad...