STMicroelectronics/STMems_Standard_C_drivers #22: lsm6dsl_pin_int1/2_route_set overrides global interrupts_enable flag

mickael9, 2018-12-05:
Take this example :

```c
    // Set INT1 to wake up
    lsm6dsl_int1_route_t int1 = {
        .int1_wu = 1,
    };
    lsm6dsl_pin_int1_route_set(&dev_ctx, int1);
    
    // Disable INT2
    lsm6dsl_int2_route_t int2 = {0};
    lsm6dsl_pin_int2_route_set(&dev_ctx, int2);
```

The call to `lsm6dsl_pin_int2_route_set` sets `TAP_CFG.INTERRUPTS_ENABLE` to 0, which globally disables all interrupts, including the INT1 one

Possible fixes :
 - Merge `lsm6dsl_pin_int1_route_set` and `lsm6dsl_pin_int2_route_set` into a single function
 - Set `TAP_CFG.INTERRUPTS_ENABLE` if one interrupt is enabled, but never clear it

This bug seems to apply to the following drivers :
 - iis2dlpc
 - ism330dlc
 - lis2dw12
 - lsm6dsl
 - lsm6dsm
 - lsm6dso

albezanc, 2018-12-06:
Hi mickael9,
I just release a new version of lsm6dsl driver that also take in account your point.
I leave this issue open till every device drivers is fixed. 
Thanks very much for your contribution.

albezanc, 2018-12-14:
The issue is now fixed on all devices.
Thanks again for your contribution!

closed 2018-12-14 by albezanc