STMicroelectronics/STMems_Standard_C_drivers #32: MISRA C compliance

LoriBru, 2019-03-28:
Are the drivers on this repository MISRA C compliant?

I see there are some references to MISRA C compliance within some header files, but there is no statement related to that on the documents/resources.
How do the drivers relate to MISRA C on a general level?

I think this should be better specified somewhere.

albezanc, 2019-04-01:
An activity concerning the coverage of MISRA C: 2012 (MISRA C3) compliance is on-going on all drivers. This activity will be completed within June 2019. 

You can found the detailed description of MISRA C: 2012 (MISRA C3) [here](https://www.misra.org.uk/Publications/tabid/57/Default.aspx).

**At the moment (01-April-2019) the only drivers not MISRA compliant are: lsm303ah, lis2ds12, iis3dhhc.**

Please note that only drivers are compliant with MISRA C: 2012 (examples are not fully compliant).

Thank you for opportunities to clarify.

LoriBru, 2019-04-02:
Thank you for the very detailed answer.
This is some really useful info!

closed 2019-04-04 by albezanc