STMicroelectronics/STMems_Standard_C_drivers #33: ISM330DLC with Arduino

isZumpo, 2019-04-19:
Hello, I am having some issues connecting an Arduino to a ISM330DLC(connected through a STEVAL-MKI182V2) using these drivers. When running the driver the device id is always 255. This is my current code:

```c
static int32_t platform_write(void *handle, uint8_t Reg, uint8_t *Bufp, uint16_t len) { 
  SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE3));
  digitalWrite(SSpin, LOW);
  SPI.transfer(Reg);
  SPI.transfer(Bufp);
  digitalWrite(SSpin, HIGH);
  SPI.endTransaction();
  return 0;
}
static int32_t platform_read(void *handle, uint8_t Reg, uint8_t *Bufp, uint16_t len) {
  Reg |= 0x80; //Set bit 0 (MSB) in order to read
  SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE3));
  digitalWrite(SSpin, LOW);
  SPI.transfer(Reg);
  *Bufp = SPI.transfer(0x00); //Dummy send to get value from module
  digitalWrite(SSpin, HIGH);
  SPI.endTransaction();
  return 0;
}
void setup() {
  delay(100);
  Serial.begin(9600);
  pinMode(SSpin, OUTPUT);
  digitalWrite(SSpin, HIGH);

  /* Initialize SPI */
  SPI.begin();
  ....
  ....
```

Does anyone know what might be wrong?

albezanc, 2019-04-24:
Hi isZumpo, 

I'm not an expert of arduino libraries, but the call: 
`SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE3));`
seems that initialize the SPI clock at [14MHz](https://www.arduino.cc/en/Reference/SPISettings). 

The ISM330DLC maximum SPI clock ferquency is 10MHz  [(Ref.  DS12171 - Rev 3 "Table 5" )](https://www.st.com/resource/en/datasheet/ism330dlc.pdf).

You can also find some examples of Arduino integration on [this project](https://github.com/stm32duino/)

isZumpo, 2019-04-26:
Thank you for the help! I have tried a few different SPI clock frequencies but none of them seem to work. Might be something wrong with my device or cables. So I bought a different device which seem to work fine.

closed 2019-04-26 by isZumpo