STMicroelectronics/STMems_Standard_C_drivers #38: About  Unico

KiLuYa, 2019-06-21:
I use your sensors in my board.
I want to use Unico to capture, process, display sensor data.
I may need to implement the communication protocol via serial port.
Do you have some document about how to do it?

By the way, I just need Unico to receive and handle data from MCU,  I don't need it send command to MCU, do you have a more light weight solution?

albezanc, 2019-06-21:
Dear KiLuYa, 

the right place for *Unico GUI* support is at this [here](https://my.st.com/ols). In any case below a brief clarification about the *Unico GUI* working modes:

- online mode designed for Professional MEMS board [STEVAL-MKI109V3](https://www.st.com/en/evaluation-tools/steval-mki109v3.html). If you want to use your board in this mode you have to implement the communication protocol described [here](https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/c6/b4/6a/03/c1/6f/46/13/DM00331058/files/DM00331058.pdf/jcr:content/translations/en.DM00331058.pdf) on your board.

- offline, without any boards. In this modality it is possible to load log files formatted as *Unico GUI* without any boards.   

You can find additional documentation about *Unico GUI* on [Unico GUI User Manual](https://www.st.com/content/ccc/resource/technical/document/user_manual/b0/8d/9f/be/cc/2c/4e/2a/CD00297387.pdf/files/CD00297387.pdf/jcr:content/translations/en.CD00297387.pdf) 

Sincerely

closed 2019-06-21 by albezanc

KiLuYa, 2019-06-21:
Great. Thanks.