STMicroelectronics/STMems_Standard_C_drivers #41: LIS3DH - Error in C-Driver-MEMS

awneil, 2019-07-19:
See: https://community.st.com/s/question/0D50X0000B6QVmzSQG/lis3dh-error-in-cdrivermems

The LIS3DH Temperature Sensor is connected to ADC3 

But lis3dh_temperature_raw_get() reads ADC1:

```C
 /**
  * @brief  Temperature output value.[get]
  *
  * @param  ctx      read / write interface definitions
  * @param  buff     buffer that stores data read
  * @retval          interface status (MANDATORY: return 0 -> no Error)
  *
  */
int32_t lis3dh_temperature_raw_get(lis3dh_ctx_t *ctx, uint8_t *buff)
{
  int32_t ret;
  ret = lis3dh_read_reg(ctx, LIS3DH_OUT_ADC1_H, buff, 1);   // <<<<< HERE !!
  return ret;
} 
```

 And why is it only reading the High byte?

albezanc, 2019-07-22:
Of course, the issue should have been resolved with the last commit.

Thank you for your contribution.

awneil, 2019-07-25:
The latest commit has:
``` C
int32_t lis3dh_temperature_raw_get(lis3dh_ctx_t *ctx, uint16_t *buff)
{
  int32_t ret;
  uint8_t dummy[2];
  ret = lis3dh_read_reg(ctx, LIS3DH_OUT_ADC3_L, dummy, 2);
  *buff = dummy[1];
  *buff = *buff * 256;
  *buff += dummy[0];
  return ret;
}
```
Shouldn't the result be signed?

albezanc, 2019-07-31:
Sure, a signed integer is certainly more consistent, we will arrange the driver. 

Thank you

closed 2019-08-02 by albezanc