STMicroelectronics/STMems_Standard_C_drivers #48: lis2dtw12 wrong Sensitivity

juttTechnology, 2019-09-25:
Hi.
    in lis2dtw12_reg.c file you are using very different Sensitivity. for example in lis2dtw12_from_fs8_lp1_to_mg function you are using 0.244 Sensitivity. according to the datasheet, it should be 0.976 or 1.952 according to the power mode.

thank you

albezanc, 2019-09-26:
Hi juttTechnology, 
the sensitivity values used in the driver take also in account the right the left aligned register format:

Supposing you are using *low power mode 1* with *full scale at 8g* 
from [Device datasheet](https://www.st.com/resource/en/datasheet/lis2dtw12.pdf) 
- sensitivity is 3.904 *mg/digit* 
- data resolution are 12 bit left aligned in 16 bit format

the x axis value in *digit* is the composition of these output registers right shifted by 4:
 (OUT_X_H << 8  + OUT_X_L) >> 4 

the *mg* value can be obtained multiplying the digit value by the sensitivity 3.904 *mg/digit*
(  ( OUT_X_H << 8  + OUT_X_L ) >> 4 ) * 3.904 = 
(  ( OUT_X_H << 8  + OUT_X_L ) / 16 ) * 3.904 =
(  ( OUT_X_H << 8  + OUT_X_L ) / 16 ) * 3.904 =
(  ( OUT_X_H << 8  + OUT_X_L ) * 0.244

thanks for the clarification opportunity

juttTechnology, 2019-09-26:
thank you for the detailed answer.

closed 2019-09-26 by juttTechnology