STMicroelectronics/STMems_Standard_C_drivers #50: Why I am stuck in I2C_ISR_TXIS flag set?

wounder123, 2019-10-03:
I have MPU6050 accerlometer, i need to interface with stm32f0 (via I2C).

Firstly i am check if the sensor is responding by reading the

"WHO_AM_I (0x75)" Register.

If the sensor responds with 0x68, this means it’s available and good to go.

but the code is stuck in while(I2C_GetFlagStatus(I2C1, I2C_ISR_TXIS) == RESET){} ,

why this happens?

__inline static void stm32_I2cSetup(void)
{
  GPIO_InitTypeDef  GPIO_InitStructure;
	I2C_InitTypeDef  I2C_InitStructure;
  
#if	_I2C1_ENABLE__
  /* Configure the I2C clock source. The clock is derived from the HSI */
  RCC_I2CCLKConfig(RCC_I2C1CLK_HSI);
    
  /* I2C_SCL_GPIO_CLK and I2C_SDA_GPIO_CLK Periph clock enable */
  RCC_AHBPeriphClockCmd(I2C1_SCL_GPIO_CLK | I2C1_SDA_GPIO_CLK, ENABLE);
  
  /* I2C Periph clock enable */
  RCC_APB1PeriphClockCmd(I2C1_CLK, ENABLE);
  
  /* Connect PXx to I2C_SCL*/
  GPIO_PinAFConfig(I2C1_SCL_GPIO_PORT, I2C1_SCL_SOURCE, I2C1_SCL_AF);
  
  /* Connect PXx to I2C_SDA*/
  GPIO_PinAFConfig(I2C1_SDA_GPIO_PORT, I2C1_SDA_SOURCE, I2C1_SDA_AF);
  
  /* GPIO configuration */  
  /* Configure I2C pins: SCL */
  GPIO_InitStructure.GPIO_Pin = I2C1_SCL_PIN;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_2MHz;
  GPIO_InitStructure.GPIO_OType = GPIO_OType_OD;
  GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP;
  GPIO_Init(I2C1_SCL_GPIO_PORT, &GPIO_InitStructure);
  
  /* Configure I2C pins: SDA */
  GPIO_InitStructure.GPIO_Pin = I2C1_SDA_PIN;
  GPIO_Init(I2C1_SDA_GPIO_PORT, &GPIO_InitStructure);	
	
  /* I2C configuration */
  /* sEE_I2C configuration */
  I2C_InitStructure.I2C_Mode = I2C_Mode_I2C;
  I2C_InitStructure.I2C_AnalogFilter = I2C_AnalogFilter_Enable;
  I2C_InitStructure.I2C_DigitalFilter = 0x00;
  I2C_InitStructure.I2C_OwnAddress1 = 0x00;
  I2C_InitStructure.I2C_Ack = I2C_Ack_Enable;
  I2C_InitStructure.I2C_AcknowledgedAddress = I2C_AcknowledgedAddress_7bit;
  I2C_InitStructure.I2C_Timing = I2C1_TIMING;
  
  /* Apply sEE_I2C configuration after enabling it */
  I2C_Init(I2C1, &I2C_InitStructure);
   
  /* sEE_I2C Peripheral Enable */
  I2C_Cmd(I2C1, ENABLE);
  
//  /* Select the EEPROM address */
// SlaveAddr = 0x60;  
	Current_I2C	 = I2C1; 
#endif

#if _I2C2_ENABLE__

#endif
}

void reg_read(void)
{
uint32_t Data = 0;
printk("~11");

 while(I2C_GetFlagStatus(I2C1, I2C_ISR_BUSY) != RESET){;;}
  printk("~22");
 I2C_TransferHandling(I2C1,0x68, 1, I2C_Reload_Mode,       I2C_Generate_Start_Write);  //I2C_SoftEnd_Mode
  printk("~33");

 while(I2C_GetFlagStatus(I2C1, I2C_ISR_TXIS) == RESET){}
 printk("~44");
 I2C_SendData(I2C1,0x75);
while(I2C_GetFlagStatus(I2C1, I2C_ISR_TC) == RESET){} 

 I2C_TransferHandling(I2C1, 0x68, 1,          I2C_AutoEnd_Mode,I2C_Generate_Start_Read);

 while(I2C_GetFlagStatus(I2C1, I2C_ISR_RXNE) == RESET) { }

Data= I2C_ReceiveData(I2C1);

while(I2C_GetFlagStatus(I2C1, I2C_ISR_STOPF) == RESET) {}

I2C_ClearFlag(I2C1, I2C_ICR_STOPCF);

 memset(ucGBuff1,0,sizeof(ucGBuff1));
 sprintf((char *)ucGBuff1,"C:  %d",Data);
printk("~22");
printk((char *)ucGBuff1);

  }

albezanc, 2019-10-04:
Dear wounder123,

in this repo **STMicroelectronics** provide C-drivers, and support on these, **ONLY for STMicroelectronics Sensors**, for this reason I'm not able to provide support on MPU6050.

So if you want to use this (*in my opinion, wonderful*) resources you have to migrate your project on a [STMicroelectronics-Inertial Modules](https://www.st.com/en/mems-and-sensors/inemo-inertial-modules.html#overview).  :-)

Concerning your **STM32** microcontroller, my suggestion is to use [CubeMX](https://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-configurators-and-code-generators/stm32cubemx.html) tool with the [STM32CubeF0](https://www.st.com/content/st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-mcu-mpu-packages/stm32cubef0.html) library for configure you microcontroller and initialize your code.

In any case the reference web page about **STMicroelectronics** products support is [this](https://www.st.com/content/st_com/en/support/support-home.html).

thanks for the clarification opportunity

wounder123, 2019-10-04:
ok..thank a lot

closed 2019-10-04 by albezanc