STMicroelectronics/STMems_Standard_C_drivers #58: LIS2DW12: Wakeup example generates a pulse on INT1 after enabling the feature

masterbar, 2020-03-06:
Hello,

I'm trying the wake-up example of the LIS2DW12. It's working except that the accelerometer generates a pulse on the INT1 pin ~1ms after the setup (i.e. after enabling the wake-up in CTRL4).
It's systematic, even if the accelerometer is motionless. The pulse duration is 5ms.
If I add some delay, few milliseconds, before configuring CTRL4, then the pulse disappear. But it's something I'd like to avoid, I would prefer to understand why the chip generates the pulse after enabling the wake-up feature. This pulse is a wrong information coming from the LIS2DW12.

Thank you.

albezanc, 2020-03-06:
Hi Greg B.,

the pulse is due to a high pass filter settling time. In the condition of the example the filter settling time is 1 sample.  

Please, refer to application note **AN5038** section **3.4 Accelerometer bandwidth** table **Table 13. Bandwidth: high-pass path** for detailed informations.

thanks for the opportunity for clarification

masterbar, 2020-03-06:
OK, thank you for the explanation @albezanc . I'll wait a little bit (1/ODR) before reading INT1.

closed 2020-03-06 by masterbar