STMicroelectronics/STMems_Standard_C_drivers #71: Iis328dq spi read data problem

nagaarjunanv, 2020-07-13:
What is the procedure to read data from iis328dq sensor 
By datasheet format isn't responding
Simply sending to read WHO_I_AM register 
0x8F is it ok or want to send another other form or want to add any other previous steps

albezanc, 2020-07-16:
hi nagaarjunanv,

if you are looking about an example of how to get data using the iis328dq driver of this repository you can refer to  [iis328dq_read_data_polling.c](https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/iis328dq_STdC/example/iis328dq_read_data_polling.c) and consider the *STEVAL_MKI109V3* case that implement an example of the SPI communication using *STM32 CubeMX*.

For any other request please refer to [ST Community Forum](https://community.st.com/s/) or [ST Support Center](https://my.st.com/ols#/ols/)

Thank you for the clarification opportunity

closed 2020-07-24 by albezanc

nagaarjunanv, 2020-10-05:
I am working in Renesas E2studio smart configuration