STMicroelectronics/STMems_Standard_C_drivers #73: LIS2DW12 Software Trigger Question

MC123T, 2020-07-17:
Hey There, 
ive got a question related to sowtware Triggering the LIS2DW12 Sensor. 

>       for (int i=0; i<0x3FF; i++);
      lis2dw12_data_rate_set(&dev_ctx, LIS2DW12_XL_SET_SW_TRIG);

what purpose has the loop there? i dont get it ..
the code is taken from  the lis2dw12_read_data_single example

albezanc, 2020-07-24:
Hi MC123T,

when the device ends a "Single data conversion" and data ready signal rise, you can read the data immedialy but you can not trigger a new conversion before 175us.

Summarizing: if you are using I2C or slow SPI the delay loop is not need because the data read operation take more than 175us, otherwise you have to add a delay loop in order to avoid to set a new trigger before 175us.

Thanks for the clarification opportunity

MC123T, 2020-07-24:
Got it! thank you very much!
That has nothing to do with the T_on time mentioned in the Datasheet right?
[image]

closed 2020-07-24 by MC123T

albezanc, 2020-07-24:
You are right, this point is not emphasized in this image.