STMicroelectronics/STMems_Standard_C_drivers #76: FIFO decompressor bug

vnerozinea, 2020-08-17:
https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/5198b93f651d9bf0270e65ebc64b0639495d4254/_resources/FIFO_decompression_utility/fifo_utility.c#L870

You are masking and rewriting "decode_temp" in a loop. It looks like after first "j" content is useless. 
Is it a bug or I am missing something?

albezanc, 2020-08-31:
Hi vnerozinea,

the role of `j` index is scrolls the 5 bits fields that represent the difference with previous data for each axis.

Concerning the *compressed data format* please consult the *application note* of your device at section
"**FIFO compression**" in particular table "**3xC compressed data output data format in FIFO**" and the numerical example "**FIFO compression example**".

Thanks for clarification opportunity

vnerozinea, 2020-08-31:
Ok, so it means that it is a bug.

On `i` you are reading `decode_temp`, 
than on `j = 0` you are masking it and storing back to `decode_temp` `decode_temp &= ((uint32_t)0x1FU << (5U * j) )`
and  for all other `j` (j == 1 and j == 2) `temp` always equals 0, which is wrong. 
`for (uint8_t j = 0; j < 3U; j++)` is useless.

albezanc, 2020-08-31:
Sorry, at first I didn't understand your point.
You are right, this is a bug.

Thank you so much for your contribution!

closed 2020-09-04 by albezanc