STMicroelectronics/STMems_Standard_C_drivers #92: lsm6dsox_ln_pg_write does not work properly

bogdanm, 2020-10-16:
HEAD is at https://github.com/STMicroelectronics/STMems_Standard_C_drivers/commit/740df6f17396a80ba3b7f9fa3e272df67ccc1ad4 when I'm writing this.

This part of `lsm6dsox_ln_pg_write` is wrong:

https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lsm6dsox_STdC/driver/lsm6dsox_reg.c#L1798-L1817

The issues there:

- The test in https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lsm6dsox_STdC/driver/lsm6dsox_reg.c#L1803 is made before the LSB is incremented
- According to your own AN5273, section 8 ("FSM configuration example"), PAGE_SEL must be written to 1 after writing all the programs, not after each byte write.

A replacement for the code above that seems to work fine for me is this:

```
  if (ret == 0) {
    for (i = 0; ( (i < len) && (ret == 0) ); i++) {
      ret = lsm6dsox_write_reg(ctx, LSM6DSOX_PAGE_VALUE, &buf[i], 1);
      lsb ++;
      /* Check if page wrap */
      if ( (lsb == 0x00U) && (ret == 0) ) {
        msb++;
        ret = lsm6dsox_read_reg(ctx, LSM6DSOX_PAGE_SEL, (uint8_t*)&page_sel, 1);
        if (ret == 0) {
          page_sel.page_sel = msb;
          page_sel.not_used_01 = 1;
          ret = lsm6dsox_write_reg(ctx, LSM6DSOX_PAGE_SEL,
                                  (uint8_t*)&page_sel, 1);
        }
      }
    }
  }
  page_sel.page_sel = 0;
  page_sel.not_used_01 = 1;
  ret = lsm6dsox_write_reg(ctx, LSM6DSOX_PAGE_SEL, (uint8_t*) &page_sel, 1);
```

The deeper issue here is that this error makes me wonder if you test your code at all. You have an example of FSM usage at https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lsm6dsox_STdC/example/lsm6dsox_fsm.c. I'm quite certain that if you tried to test that code with the fauly function above, it would have never worked. Well, in this particular case it wouldn't even compile, because `lsm6dsox_enbedded_sens_set` in https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lsm6dsox_STdC/example/lsm6dsox_fsm.c#L270 should obviously have been called `lsm6dsox_embedded_sens_set` (notice the typo in `enbedded`). So you don't even try to **compile** your code before you make it public?

In any case, many thanks for forcing me to look for an error that was not in my code to begin with.

albezanc, 2020-10-16:
I can confirm that all the examples have been verified even if it is evident that the page rollback condition in the `_ln_pg_write` function has never been used and thank you for the tip.

We work continuously to try to facilitate our users, in this perspective about a week ago 80b4f964acb2dcb213c39e849372a212e2747a6b we have updated some examples. Unfortunately, a gross error has escaped the flow of control.

I offer my deepest apologies for what happened

closed 2020-10-16 by albezanc

bogdanm, 2020-10-16:
Thank you for your kind words and for acknowledging this issue. However, I don't believe that you need a page rollback condition to trigger this bug. The start address for FSM programs is 0x0400, which corresponds to `msb==4, lsb==0` in your code: https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/740df6f17396a80ba3b7f9fa3e272df67ccc1ad4/lsm6dsox_STdC/driver/lsm6dsox_reg.c#L1770-L1771. Then, on the first iteration of the "for" loop https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/740df6f17396a80ba3b7f9fa3e272df67ccc1ad4/lsm6dsox_STdC/driver/lsm6dsox_reg.c#L1798 the condition https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/740df6f17396a80ba3b7f9fa3e272df67ccc1ad4/lsm6dsox_STdC/driver/lsm6dsox_reg.c#L1803 is true and thus the msb is incremented: https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/740df6f17396a80ba3b7f9fa3e272df67ccc1ad4/lsm6dsox_STdC/driver/lsm6dsox_reg.c#L1805 and consequently the page number is immediately changed: https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/740df6f17396a80ba3b7f9fa3e272df67ccc1ad4/lsm6dsox_STdC/driver/lsm6dsox_reg.c#L1810-L1811 In other words, it looks like the code writes the first byte of the FSM program(s), then immediately increments the page number, thus skipping all the other bytes in the previous page. This is why I said that I don't see how this code ever worked.

In any case, I appreciate the quick reaction and the fix.