STMicroelectronics/STMems_Standard_C_drivers #98: Changing I2C Address for LSM9DS1

hamadbinghalib, 2020-11-12:
Hi. First of all just wanted to say thanks for making these drivers. I'm coming from old MEMS code for LSM6 and been using it on the LSM9DS1. Decided to migrate to these drivers.

I ran you driver by default, and I couldn't communicate over I2C. After some digging and comparing to my old code, I noticed that platform_read and platform_write are using the same I2C address from sensbus imu_bus in both example codes.

The default comes from LSM9DS1_IMU_I2C_ADD_H which is 0xD7. For both read and write according the poll example code.

When I changed the write address to D4 and read to D5, it worked for IMU. Didn't try the MAG but I assume its the same issue. 

I don't think its a bug, but maybe a difference in configuration. Could someone please explain why my setup is different from yours? 

Thanks in advance.

albezanc, 2020-11-13:
Hi hamadbinghalib,

Thank you for appreciating our effort in satisfying the users of our devices! The examples were tested using [STEVAL-MKI159V1](https://www.st.com/en/evaluation-tools/steval-mki159v1.html) on a breadboard for connections with [NUCLEO-F411RE](https://www.st.com/en/evaluation-tools/nucleo-f411re.html). 
Referring to [LSM9DS1 datasheet Rev3](https://www.st.com/resource/en/datasheet/lsm9ds1.pdf) table 19 and table 20, the polarization of the pins 5 and 6 can change the I2C address: 

- *the pins 5 (SDO_A/G)*: 
  - high LSM9DS1_IMU_I2C_ADD_H (Read=D7-Write=D6) 
  - low LSM9DS1_IMU_I2C_ADD_L (Read=D5-Write=D4)
- *the pins 6 (SDO_M)*:
  - high LSM9DS1_MAG_I2C_ADD_H (Read=3D-Write=3C) 
  - low LSM9DS1_MAG_I2C_ADD_L (Read=39-Write=38)

The [STM32CubeF4 I2C driver](https://github.com/STMicroelectronics/STM32CubeF4/)  internally manage the R/W bit through the macros `2C_7BIT_ADD_READ` and `I2C_7BIT_ADD_WRITE`, so only 8bit I2C read address is need.

Concluding the differencies consists in the polarizzation of pins 5 and 6.

Thanks for the opportunity for clarification.

hamadbinghalib, 2020-11-14:
Thanks for the reply. Very clear now.

closed 2020-11-14 by hamadbinghalib