STMicroelectronics/STMems_Standard_C_drivers #107: ISM330DHCX - CTRL9_XL DEVICE_CONF

AxelSchmidtDD, 2021-01-22:
### ISM330DHCX

### Question

What is the meaning of CTRL9_XL bit 1 "DEVICE_CONF"? In the data sheet is written "It is recommended to always set this bit to 1 during device configuration.", but in STM's demo applications (STMems_Standard_C_drivers/ism330dhcx_STdC/example/) this register is not used at all.

albezanc, 2021-01-25:
Hi, the handling of **ctrl8_xl.device_conf** bit is missing on all the examples and it is a bug even if the examples seems work correctly.

This issue will be fixed within this week.

The solution will be to set the ctrl8_xl.device_conf bit to 1 just outside the ism330dhcx_reset_get control loop. The ctrl8_xl.device_conf bit will never be cleared.

Thanks very much for your contribution!

albezanc, 2021-01-29:
Thank you for your contribution!

closed 2021-01-29 by albezanc