STMicroelectronics/STMems_Standard_C_drivers #108: Example for lis2de12_fifo_status_get with LIS2DE12

capiman, 2021-01-24:
### Device part numbers

LIS2DE12

### Summary

I want to use FIFO to receive raw X/Y/Z values. Do you perhaps have an example how to setup sensor
to get values from INT1 and especially how interrupt routine shall look like?

### Basic example

Add example

### Motivation

I need the FIFO because I want to save power and put microcontroller to sleep till sensor has collected up to 32 raw X/Y/Z values.
Then I want to wake up microcontroller and get all the values in one go.

capiman, 2021-01-24:
There seems to be a ompareable example in example folder of LIS2DS12

https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lis2ds12_STdC/example/lis2ds12_multi_read_fifo.c

capiman, 2021-01-25:
But I think there is perhaps an error?

lis2dh12_fifo_watermark_set set field FTH. Field FTH is 5 bits, can go from 0 to 0x1F.

But example uses this code:

/* Define number of byte for each sensor sample */
#define OUT_XYZ_SIZE    6

/* Define FIFO watermark to 10 samples */
#define FIFO_WATERMARK    OUT_XYZ_SIZE * 10

...

  /* Set FIFO watermark to FIFO_WATERMARK */
  lis2ds12_fifo_watermark_set(&dev_ctx, FIFO_WATERMARK);
  
  Is multiplication with OUT_XYZ_SIZE correct?

capiman, 2021-01-25:
Same here?

      /* Read number of sample in FIFO */
      lis2ds12_fifo_data_level_get(&dev_ctx, &num);
      num_pattern = num / OUT_XYZ_SIZE;

But there is more calculation behind. It looks like LIS2DS12 is not comparable to LIS2DE12, at least in this point?

#define LIS2DS12_FIFO_THS              0x2EU
typedef struct {
  uint8_t fth                 : 8;
} lis2ds12_fifo_ths_t;

Can't find 8 bit field in the version of datasheet of LIS2DE12, at least the version I have.

Assume separate example for FIFO and LIS2DE12 is needed...

albezanc, 2021-01-25:
Hi capiman,

please don't mix devices, their behavior can be very different.
A FIFO reading example with lis2de12 will be added this week.

Thanks very much for your suggestion.

capiman, 2021-01-26:
PS: AN5005 - Page 46 - DocID030302 Rev 2: Chapter 9.3.2 -> There is mentioned CTRL_REG5. I think it must be CTRL_REG3 instead.

albezanc, 2021-01-29:
Thank you very much you for all your feedbacks! The examples added should be match your needs

closed 2021-01-29 by albezanc