STMicroelectronics/STMems_Standard_C_drivers #121: const correctness for platform write

pabigot, 2021-04-15:
Some development efforts have a policy that when a function promises not to modify data referenced by pointer, the referenced data should have a `const` qualifier in the function parameter declaration.

In that situation the declaration for `stmdev_write_ptr` may force the caller to cast away const to satisfy the current signature:

```
typedef int32_t (*stmdev_write_ptr)(void *, uint8_t, uint8_t *,
                                    uint16_t);
```

This casting would not be necessary if the signature were instead:
```
typedef int32_t (*stmdev_write_ptr)(void *, uint8_t, const uint8_t *,
                                    uint16_t);
```

It doesn't seem reasonable to expect any platform write implementation to be allowed to change data that it is transferring to the device, so tightening the declaration should be technically OK.  But since the change would affect every existing platform_write implementation it would break existing code, so should probably be part of an update to version 2.x.

Just something to keep in mind.

pabigot, 2021-04-19:
As noted in https://github.com/zephyrproject-rtos/zephyr/pull/34389 this would also apply to the `stmdev_ctx_t *ctx` argument passed to all driver functions.  That should be `const stmdev_ctx_t *ctx` since the driver functions shouldn't be modifying its content.

avisconti, 2021-04-20:
> As noted in [zephyrproject-rtos/zephyr#34389](https://github.com/zephyrproject-rtos/zephyr/pull/34389) this would also apply to the `stmdev_ctx_t *ctx` argument passed to all driver functions. That should be `const stmdev_ctx_t *ctx` since the driver functions shouldn't be modifying its content.

Yes, that's correct. I confirm that it would be nice to add the 'const' to the fields. I already discussed internally, and if that would be changed in the STdC drivers I will align them in zephyr and would be able to remove the bad casting I had to insert when using the ctx handle.

closed 2021-05-31 by albezanc