STMicroelectronics/STMems_Standard_C_drivers #146: More typos -> LSM6DSO

capiman, 2021-10-12:
I was initially not aware that each directory is  for one sensor.
I thought it is a software variant and even wanted to question what is different between each directory, e.g.
like LSM6DSO, LSM6DSOX, LSM6DSO32, LSM6DSO32X.

Question beside: Are they in some kind (backward) compatible?
So e.g. code for LSM6DSO can run on all other mentioned (when not using the additional features)?

But back to LSM6DSO source code:

Minor:
https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lsm6dso_STdC/examples/lsm6dso_self_test.c#L307

LIS2DH12_GY_ST_NEGATIVE
-> 
LSM6DSO_GY_ST_NEGATIVE

https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lsm6dso_STdC/examples/lsm6dso_self_test.c#L348

lsm6dso_xl_data_rate_set(&dev_ctx, LSM6DSO_GY_ODR_OFF);
->
lsm6dso_gy_data_rate_set(&dev_ctx, LSM6DSO_GY_ODR_OFF);

(same as https://github.com/STMicroelectronics/STMems_Standard_C_drivers/issues/142 )

I am at the moment using only self test code, so not checked the other example files.

capiman, 2021-10-13:
Searching over complete repo gives following result:

marie@marieubuntu:~/STMems_Standard_C_drivers$ grep -r xl_data_rate_set * | grep -e "_GY_"
lsm6ds3tr-c_STdC/examples/lsm6ds3tr_c_self_test.c:  lsm6ds3tr_c_xl_data_rate_set(&dev_ctx, LSM6DS3TR_C_GY_ODR_OFF);
lsm6dso_STdC/examples/lsm6dso_self_test.c:  lsm6dso_xl_data_rate_set(&dev_ctx, LSM6DSO_GY_ODR_OFF);
lsm6dsrx_STdC/examples/lsm6dsrx_self_test.c:  lsm6dsrx_xl_data_rate_set(&dev_ctx, LSM6DSRX_GY_ODR_OFF);
marie@marieubuntu:~/STMems_Standard_C_drivers$

avisconti, 2021-10-19:
> I was initially not aware that each directory is for one sensor. I thought it is a software variant (or X as place holder for some hardware variants) and even wanted to question here what is different between each directory, e.g. like LSM6DSO, LSM6DSOX, LSM6DSO32, LSM6DSO32X.
> 
> Question beside: Are they in some kind (backward) compatible? So e.g. code for LSM6DSO can run on all other mentioned device (when not using the additional features)?
> 

They are somehow backward compatible, yes.

> But back to LSM6DSO source code:
>
> Minor: https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/lsm6dso_STdC/examples/lsm6dso_self_test.c#L307
>
> LIS2DH12_GY_ST_NEGATIVE -> [...]

OK, thanks!
Maybe you can try to provide again a PR, like you already did. Just mention the number of this bug with '#'