STMicroelectronics/STMems_Standard_C_drivers #147: lsm303ah: wrong register used in lsm303ah_xl_fifo_data_level_get() function

vadim-bey, 2022-03-11:
### Device part numbers

lsm303ah

### Type of bug

driver - lsm303ah_STdC/driver/lsm303ah_reg.c

### Describe the bug

function lsm303ah_xl_fifo_data_level_get() uses a wrong register to read the second part of available samples' count. Also DIFF[8] is shifted incorrectly.

From the datasheet:
FIFO_SAMPLES_A register description
DIFF [7:0]

Concatenated with DIFF8 bit in FIFO_SRC_A (2Fh) register, it represents the
number of unread samples stored in FIFO. (000000000 = FIFO empty;
100000000 = FIFO full, 256 unread samples).

### Additional context

Proposed patch:

diff --git a/lsm303ah_STdC/driver/lsm303ah_reg.c b/lsm303ah_STdC/driver/lsm303ah_reg.c
index e0a1168..a844ea1 100644
--- a/lsm303ah_STdC/driver/lsm303ah_reg.c
+++ b/lsm303ah_STdC/driver/lsm303ah_reg.c
@@ -3348,11 +3348,11 @@ int32_t lsm303ah_xl_fifo_data_level_get(stmdev_ctx_t *ctx, uint16_t *val)
   int32_t ret;
 
   ret = lsm303ah_read_reg(ctx, LSM303AH_FIFO_SRC_A, (uint8_t*)&fifo_src_a, 1);
-  if(ret == 0){
-    ret = lsm303ah_read_reg(ctx, LSM303AH_FIFO_SRC_A, &fifo_sample_a, 1);
+  if(ret == 0) {
+    ret = lsm303ah_read_reg(ctx, LSM303AH_FIFO_SAMPLES_A, &fifo_sample_a, 1);
     *val = fifo_src_a.diff;
-    *val = *val << 7;
-    *val += fifo_sample_a;
+    *val = *val << 8;
+    *val |= fifo_sample_a;
   }
   return ret;
 }

avisconti, 2022-03-11:
@vadim-bey 
Yes, the routine is bugged, thanks for pointing out.

If you agree you can submit a PR addressing this bug and I'll review and accept it. In alternative I can do the patch.
Thanks!

vadim-bey, 2022-03-11:
I trust you on creating this PR, please fix it, thanks :)