STMicroelectronics/STMems_Standard_C_drivers #148: lsm6ds3tr-c: Read and write registers error

echoming, 2022-05-07:
###lsm6ds3tr-c

lsm6ds3tr-c

### Type of bug

examples

### Describe the bug

I tried to read lsm6ds3tr-c'regsters by STM32F103C8T6 running in main frequency 72MHz. SPI 4-wire is used. But lsm6ds3tr-c return '0xFF'. 

Function "platform_read" use "HAL_SPI_Transmit"&"HAL_SPI_Receive" to send address&receive data. I rewrite Function "platform_read"use "HAL_SPI_TransmitReceive" to finish a coummunicate continuously. And the correct data is fetched.
The problem is same to write regsters.

The similar issue is find by [D.luffy](https://blog.csdn.net/liangbin414/article/details/103805836)

### Additional context

"platform_read "is rewrited like this:
static int32_t platform_read(void *handle, uint8_t reg, uint8_t *bufp,
                             uint16_t len)
{
#if defined(NUCLEO_F411RE)
  HAL_I2C_Mem_Read(handle, LSM6DS3TR_C_I2C_ADD_L, reg,
                   I2C_MEMADD_SIZE_8BIT, bufp, len, 1000);
#elif defined(STEVAL_MKI109V3)
	uint8_t temp_Tx[len+1];//temp for transmit
	uint8_t temp_Rx[len+1];//temp for transmit
	
  reg |= 0x80;
  HAL_GPIO_WritePin(CS_up_GPIO_Port, CS_up_Pin, GPIO_PIN_RESET);
//  HAL_SPI_Transmit(handle, &reg, 1, 1000);
//  HAL_SPI_Receive(handle, bufp, len, 1000);

	temp_Tx[0]=reg;
	HAL_SPI_TransmitReceive(handle,temp_Tx,temp_Rx,len+1,1000);
	for(int i=0;i<len;i++)
	{
		bufp[i]=temp_Rx[i+1];
	}
  HAL_GPIO_WritePin(CS_up_GPIO_Port, CS_up_Pin, GPIO_PIN_SET);
#elif defined(SPC584B_DIS)
  i2c_lld_read(handle, LSM6DS3TR_C_I2C_ADD_L & 0xFE, reg, bufp, len);
#endif
  return 0;
}

Padifu03, 2022-09-20:
Hi echoming,
I have been working on LSM6DS3TR-C with the uC STM32L431RCT6. I had the same bug as you but, since a time reading the datasheet and comparing registers with the driver´s ones, I fixed it.
The bug is related to the slave address(SAD) and the SA0 pin, in my case the SA0 pin is connected to the supply voltage so my SAD is D6h to write and D7h to read from slave.
In the driver it appears that D5h is the write register from slave which is wrong and that made that the code doesn´t work.
**Additional context**
File: lsm6ds3tr-c_reg.h

Code of driver:
/** I2C Device Address 8 bit format  if SA0=0 -> D5 if SA0=1 -> D7 **/
#define LSM6DS3TR_C_I2C_ADD_L     0xD5U
#define LSM6DS3TR_C_I2C_ADD_H     0xD7U

Code of driver changed:
/** I2C Device Address 8 bit format  if SA0=1 -> D6 (W) if SA0=1 -> D7 (R)**/
#define LSM6DS3TR_C_I2C_ADD_H_W     0xD6U	
#define LSM6DS3TR_C_I2C_ADD_H_R      0xD7U
/** I2C Device Address 8 bit format  if SA0=0 -> D4 (W) if SA0=0 -> D5 (R)**/
#define LSM6DS3TR_C_I2C_ADD_L_W     0xD4U	
#define LSM6DS3TR_C_I2C_ADD_L_R      0xD5U

rinscr3003, 2024-07-23:
Please take a check about your SPI clock frequency. This chip doesn't work when frequency is above ~12MHz.

Padifu03, 2024-07-23:
Hi @rinscr3003 that could be the case, in my case I realized that my board has a problem with SA0 and had a fluctuating value between 1 and 0. The registers defined in the macros by STM were ok since the methods of writing and reading take care of the read/write bit and complete the register definition which is a 7 bits one.
Anyway, thanks!

avisconti, 2024-10-10:
It seems to me that the issue is not related to the STdC code and I'm closing it. If there are still problems or questions pls feel free to reopen it. Thx.