STMicroelectronics/STMems_Standard_C_drivers #156: Replace sprintf with snprintf

Hadatko, 2023-01-26:
### Device part numbers

-

### Summary

Brief explanation of the feature.

Hello, newer compilers are starting to complain related to usage of sprintf function (clang returning warning about deprecation). Their are suggesting to replace that call with snprintf. May you consider align your code?

### Basic example

### Motivation

```sh
'sprintf' is deprecated: This function is provided for compatibility reasons only. Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.
```

avisconti, 2024-10-10:
Fixed by faefa8d

closed 2024-10-10 by avisconti

Hadatko, 2024-10-14:
Funny i was on training from ST at that time 😂