STMicroelectronics/STMems_Standard_C_drivers #159: lsm6dsox does not build when optimization level is O2

idea--list, 2024-01-01:
### Device part numbers

lsm6dsox, not sure about other sensors

### Type of bug

Build error.

### Describe the bug

Project does not build if compiler optimization level is O2.

### Additional context

I integrated the driver into an ESP-IDF project of mine. The project builds fine if optimization level is Og, but fails to build when trying to build at level O2.

[image]

mu578, 2024-01-02:
From that, quick take, without looking, a memset taken out?

avisconti, 2024-01-02:
Can you pls try to remove the ctx test when ctrl5_c initialization with bytecpy()?

```
  /* reading registers to be configured */
  if (ctx != NULL)
  {
```

I need to move the ``` (ctx != NULL)``` test inside the read_reg and write_reg.
If this solves I would do a commit.
Thanks

avisconti, 2024-01-02:
Can you try to initalize ctrl5_c (and others) in lsm6dsox_mode_set()?

```
  lsm6dsox_ctrl5_c_t ctrl5_c = {0};
  lsm6dsox_ctrl6_c_t ctrl6_c = {0};
  lsm6dsox_ctrl7_g_t ctrl7_g = {0};
```
If this solves, I will submit a new commit

idea--list, 2024-01-02:
Seems if i slightly modify these 2 lines at the beginning of `lsm6dsox_mode_set` function like
```
lsm6dsox_spi2_ctrl2_ois_t spi2_ctrl2_ois = {0};
lsm6dsox_ctrl5_c_t ctrl5_c = {0};
```

Then it builds as expected.