STMicroelectronics/STMems_Standard_C_drivers #160: Part Number lsm6dsm issue

michaelboeding, 2024-01-03:
### Question

I'm trying to use the Pedometer functionality on the lsm6dsm and there doesn't seem to be an example on how to do this or a get method to actually get the step count. Am I missing something here or is it missing from the library? I seem to be getting just 0s when I try to set it up. 

https://github.com/STMicroelectronics/lsm6dsm-pid

// Set up the accelerometer //enable step count 
/* Enable pedometer algorithm */
lsm6ds3tr_c_pedo_sens_set(&dev_ctx, PROPERTY_ENABLE);

// Example: Set minimum time to 100 ms
lsm6ds3tr_c_pedo_fs_t scale = LSM6DS3TR_C_PEDO_AT_2g; // or LSM6DS3TR_C_PEDO_AT_4g
lsm6ds3tr_c_pedo_full_scale_set(&dev_ctx, scale);
// Example: Set minimum threshold to 10h
lsm6ds3tr_c_pedo_threshold_set(&dev_ctx, 0x10);
// uint8_t debounceTime = 0x0D;  // Default value: 01101
// uint8_t debounceSteps = 0x06; // Default value: 110
uint8_t debounceTime = 0x0D; // Default or custom value
uint8_t debounceSteps = 0x01; // Set to 1 for immediate step count increment
// Set Debounce Time
lsm6ds3tr_c_pedo_timeout_set(&dev_ctx, debounceTime);
// Set Debounce Steps
lsm6ds3tr_c_pedo_debounce_steps_set(&dev_ctx, debounceSteps);
// lsm6ds3tr_c_func_src1_t func_src1;
//lsm6ds3tr_c_read_reg(&dev_ctx, LSM6DS3TR_C_FUNC_SRC1, (uint8_t*)&func_src1, 1);
/* Reset step counter */
lsm6ds3tr_c_pedo_step_reset_set(&dev_ctx, PROPERTY_ENABLE);
    
    
Read the low and high parts of the step count
lsm6ds3tr_c_read_reg(&dev_ctx, LSM6DS3TR_C_STEP_COUNTER_L, &step_count_low, 1);
lsm6ds3tr_c_read_reg(&dev_ctx, LSM6DS3TR_C_STEP_COUNTER_H, &step_count_high, 1);

//print the high and low step count
printf("Step count low: %d\n", step_count_low);
printf("Step count high: %d\n", step_count_high);

// Combine them to get the full step count
step_count = (step_count_high << 8) | step_count_low;

Thanks any guidance would be appreciated.

avisconti, 2024-05-14:
why are you using the lsm6ds3tr_c APIs instead of the lsm6dsm ones? Is it just a typo here?
Anyway, you may take a look to lsm6dso pedometer and see if it is of any help.
You can also take a look at AN4987 Paragraph 6.1

michaelboeding, 2024-05-14:
Hey @avisconti thanks for the reply. I got all of this working now.

closed 2024-05-14 by michaelboeding