STMicroelectronics/STMems_Standard_C_drivers #161: "threshold" is misspelled as "treshold" in many drivers

arbrauns, 2024-03-11:
### Device part numbers

```
$ rg --files-with-matches treshold | cut -d/ -f1 | sort -u
a3g4250d_STdC
ais328dq_STdC
ais3624dq_STdC
h3lis100dl_STdC
h3lis331dl_STdC
i3g4250d_STdC
iis2mdc_STdC
iis328dq_STdC
ism303dac_STdC
lis2mdl_STdC
lis331dlh_STdC
lps22ch_STdC
lps22hh_STdC
lps27hhtw_STdC
lps27hhw_STdC
lsm303agr_STdC
lsm303ah_STdC
```

### Type of bug

driver

### Describe the bug

The word "threshold" is misspelled as "treshold" in some function names and comments.

### Additional context

This was introduced with the [very first driver in the repository](https://github.com/STMicroelectronics/STMems_Standard_C_drivers/commit/7c8c355e8cfd5332c2f76675e3a16d43b4510559) and has been copy-pasted ever since. Notably some functions in that commit were spelled correctly.

avisconti, 2024-03-19:
I'll address it in next release of STdC drivers. Thanks!

avisconti, 2024-05-14:
Fixed by 637ee8f93e0

closed 2024-05-14 by avisconti