STMicroelectronics/STMems_Standard_C_drivers #164: wrong array size in iis3dwb fifo example

chenjy96, 2024-04-18:
### Device part numbers

IIS3DWB nRF52840

### Type of bug
I wrote my program with reference to [iis3dwb fifo example](https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/iis3dwb_STdC/examples/iis3dwb_fifo.c), but the spi transfer will be stuck at the second loop. 

### Describe the bug
I check the value of `num` which is assigned  on [here](https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/iis3dwb_STdC/examples/iis3dwb_fifo.c#L174).
In the first loop, the value of `num` is equal to `FIFO_WATERMARK`, but after the second loop, the value is always 511 which is the same value of FIFO size.  So the size of array `fifo_data` which is defined [here](https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/iis3dwb_STdC/examples/iis3dwb_fifo.c#L85), should be 511, rather than `FIFO_WATERMARK`.

mu578, 2024-04-18:
511 is the max value that could be assigned, so in the example the window data is set to 256 then the cursor should move on thru the FIFO queue buffer. so that's not a size problem in itself but a mismatch logic between raw memory and processed frames.

chenjy96, 2024-04-22:
> 511 is the max value that could be assigned, so in the example the window data is set to 256 then the cursor should move on thru the FIFO queue buffer. so that's not a size problem in itself but a mismatch logic between raw memory and processed frames.

I modified [iis3dwb fifo example](https://github.com/STMicroelectronics/STMems_Standard_C_drivers/blob/master/iis3dwb_STdC/examples/iis3dwb_fifo.c) by using nRF SDK, but the value of  `num` which  represent number of unread data is not as I expected. 
In this example, value of  `num` should be the same as `FIFO_WATERMARK` when  FIFO_ WTM_IA at FIFO_STATUS2 register is 1. 
But my situation is that the value of `num`  is always 512 after the second loop.
Can you tell me the probable reason?  I'm concerned that FIFO_WTM_IA is not being triggered properly, maybe resulting in getting data that will be missing.

By the way, my read_reg and write_reg implementation is showed below.
```c
uint8_t SpiInOut (stmdev_ctx_t* spi_ctx, uint8_t outData)
{
    uint32_t err_code;
    uint8_t inData ;
    
    if ((spi_ctx == NULL) || (&(spi_ctx->spi_handle) == NULL))
    {
        APP_ERROR_CHECK_BOOL (false);
    }

    err_code = nrf_drv_spi_transfer (&(spi_ctx->spi_handle), &outData, 1, &inData, 1);
    APP_ERROR_CHECK(err_code);

    return inData;
}

int32_t SpiWriteRegisters(void* handle, uint8_t reg, const uint8_t* buffer, uint16_t len)
{
    UNUSED_PARAMETER(handle);

    GpioWrite(&iis3dwb_.spi.Nss, 0);
    SpiInOut(&iis3dwb_.spi, reg);
    for(uint16_t i = 0; i < len; i++)
    {
        SpiInOut(&iis3dwb_.spi, buffer[i]);
    }
    GpioWrite(&iis3dwb_.spi.Nss, 1);

    return 0;
}

int32_t SpiReadRegisters(void* handle, uint8_t reg, uint8_t* buffer, uint16_t len)
{
    UNUSED_PARAMETER(handle);
    
    // Set first bit to 1;
    reg |= 0x80;
    GpioWrite(&iis3dwb_.spi.Nss, 0);
    SpiInOut(&iis3dwb_.spi, reg);
    
    for(uint16_t i = 0; i < len; i++)
    {
        buffer[i] = SpiInOut(&iis3dwb_.spi, 0);
    }

    GpioWrite(&iis3dwb_.spi.Nss, 1);

    return 0;
}
```

mu578, 2024-04-22:
Hello I didn't have a chance to look into was just casting an educated guest (top of my head) on managing a `circular stream` ; Do you have any compilation warnings or so ? soemthing not initialized ? is FIFO_WATERMARK re-defined ?

closed 2024-04-22 by chenjy96